ct-Open-Source/tuya-convert #1124: AP gone after putting in wrong wifi credentials

DrR0X-glitch, 2024-01-09:
# Before you tell me, YES I read the docs that said **Triple check**

So I flashed tasmota lite, was very easy, thanks for this tool.

Issue is I typed in wifi name and then clicked checkbox looking thing next to password to turn it on and that must've submitted a blank password. So now it does not broadcast AP and my light switch doesn't work :(

Can I fix this at all OTA, or should I just swap it out and re-flash over serial?

Dunky13, 2025-01-28:
@DrR0X-glitch I've found this solution: https://tasmota.github.io/docs/Device-Recovery/#fast-power-cycle-device-recovery
Basically pull the plug 7 times in/out of the outlet, it will reset itself. Worked for me like a charm.

Adding this to the docs would be helpful. In the FAQ reference the Tasmota FAQ

DrR0X-glitch, 2025-01-29:
> @DrR0X-glitch I've found this solution: https://tasmota.github.io/docs/Device-Recovery/#fast-power-cycle-device-recovery
> Basically pull the plug 7 times in/out of the outlet, it will reset itself. [...]

I'll try this, think I already did but will check again. If it works I'll make a PR to the docs.