ct-Open-Source/tuya-convert #1131: AP starts but stops almost immediately

david00yates, 2024-06-10:
Hope this helps someone in the future as it took me a ton of time to resolve this issue.

Raspberry Pi 3B running Debian Bookworm
Linux pi 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux

I was having an issue where the vtrust-flash AP would start but almost immediately stop.

The logs for smarthack-wifi.log showed the following error:

"Failed to stop network-manager.service: Unit network-manager.service not loaded."

This turned out to be the same problem as another user reported affecting Kali Linux - Debian uses NetworkManager.service rather than network-manager.service.
Thanks to the other user's helpful guide I was able to fix the issue by editing setup_ap.sh replacing two instances of network-manager with NetworkManager.

MatzeP, 2024-06-10:
Hi,
1. can you help me where you find the log?
2. Understand I right you change the two "network-manager" to "NetworkManager"

mlossin, 2024-09-06:
I can confirm that workaround. Just change the two occurances and it works