ct-Open-Source/tuya-convert #1135: Gosund WP3 Flashing - Still usable with Tasmota?

coreyfaure, 2024-09-05:
Recently purchased some Gosund WP3s from Amazon to flash over to Tasmota.
[image]

I took one of four apart to poke around and connect VCC, GND, TX, RX, and IO0 correctly as so many guides have pointed out, but the chip I found in the WP3 modules looks very different from the guides I see online. I don't see IO0 anywhere in this device either.
[image]

Does anyone know how to handle this difference?

Scope666, 2024-09-26:
I'm in the same boat ... I have one where the Wi-Fi module has NO markings at all on it.  I know it's flashable because I've had Tasmota and ESPHome on it, but when I tried to flash back it ended up bricked.

https://github.com/ct-Open-Source/tuya-convert/issues/207#issuecomment-2094987673

divadiow, 2025-01-13:
This is a Lightning Semiconductor LN882H IoT chip. 

Opensource replacement firmware:
https://github.com/openshwprojects/OpenBK7231T_App
https://www.elektroda.com/rtvforum/topic4096854.html
https://www.elektroda.com/rtvforum/topic4028087.html

If you decide to convert please consider taking a backup first and posting that on the Elektroda forums along with your experiences with this device