ct-Open-Source/tuya-convert #1142: GHome EP2-A can not flashed with Tuya-Convert :-(

andrehamburg, 2024-10-20:
Hi guys, 
first, sorry for my bad english. Hopefully you can understand me.
I tried to flash some of the plugs, but Tuya-Convert tells me, "Your device does not use an ESP82xx". The plug are selled at amazon with flashing possibility. It has the Tuya 1.0.6 Firmware (i checked that in the Tuya SmartLife App and can confirm that) and on the Package is a Label "ESP82XX".
In the Amazon-recommondations are severel postings with a positive Tasmota flashing Feedback.
I found a tipp it in the web to add 34987A to scripts/oui/esp.txt No success :-(
I'm using a clean, fresh installed Ubuntu.

I give up. No more idea. Can anyone help?

[image]

andrehamburg, 2024-10-20:
Now i know that 34987A ist the manufacturer part of the MAC. I checked it with my plugs and 34987A is correct.  Just for information.

NickBuilder, 2024-12-10:
Hi,
I also bought a box of four smart plugs. The same model as in your photo. I had them delivered just the other day
My box has the reference/product id B09Q5BLMVT written on the bottom.

I had problem doing the tuya-convert procedure at first but I finally managed to succeed!
Some important steps:
- add the six first mac-id characters to the scripts/oui/esp.txt file
- run the ./start_flash.sh command with sudo privileges (whether or not this is important might depend on how you ran the install script) 

By not running the start_flash with sudo I got a ModuleNotFoundError, I didn't notice this at first.
Check if this is also the case for you.

```
pi@raspberrypi:~/tuya-convert $ ./start_flash.sh
tuya-convert v2.4.6
/--------/
IMPORTANT
1. Connect any other device (a smartphone or something) to the WIFI vtrust-flash
   This step is IMPORTANT otherwise the smartconfig may not work!
2. Put your IoT device in autoconfig/smartconfig/pairing mode (LED will blink fast). This is usually done by pressing and holding the primary button of the device
   Make sure nothing else is plugged into your IoT device while attempting to flash.
3. Press ENTER to continue

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Traceback (most recent call last):
  File "/home/pi/tuya-convert/scripts/./smartconfig/main.py", line 15, in <module>
    from smartconfig import smartconfig
  File "/home/pi/tuya-convert/scripts/smartconfig/smartconfig.py", line 43, in <module>
    from multicast import multicast_head, encode_multicast_body
  File "/home/pi/tuya-convert/scripts/smartconfig/multicast.py", line 12, in <module>
    from Cryptodome.Cipher import AES
ModuleNotFoundError: No module named 'Cryptodome'
........................................................................................................................
Timed out while waiting for the device to (re)connect
======================================================
```

NickBuilder, 2024-12-11:
I don't have be a problem with the tuya-convert approach. I just wanted to assist you. 🙂

As I wrote in my post:
_"I had problem doing the tuya-convert procedure at first but I finally managed to succeed!"_

But it's a good thing you solved it.
Perhaps other people can be helped by this discussion. 

Take care!

andrehamburg, 2024-12-12:
Oh damn, i hate my english :-( But you're right. Possibly it helps others.

closed 2024-12-12 by andrehamburg

wexstorm, 2025-01-15:
@NickBuilder Thanks for the tip with the mac-address. Added the first 6 digits of my GHome EP2-A Plugs (34987A) to the config file and it flashed instantly.

bwdmnn, 2025-02-05:
I just got a package of GHome EP2-A, marked with ESP82xx. So they should be Tasmota capable.
I installed everything on a Raspi3B+, connect with a LAN-cable
I see the WLAN vtrust-flash and can connect with my Phone.

The EP2-A is flashing fast.
When I run sudo ./start_flash, everything starts normal, but it always ends up with " Starting AP in a screen............."
No Cntrl-C is possible any more.
Any idea what is going wrong?
Any help is very much appreciated!

NickBuilder, 2025-02-05:
Are you sure your phone actually connects to the AP?

My phone at first wanted to reject the connection as there wasn't an internet connection behind the AP.
I then had to force the phone to connect anyway to successfully establish the wifi connection.

bwdmnn, 2025-02-05:
NickBuilder,

yes, I set the phone to Offline-Mode, activated WLAN, connected to vtrust-flash. In the browser I connected to 192.42.42.1 and got the message that I was connected.
So I think that worked, thought I had the same worries as you addressed.

bwdmnn, 2025-02-06:
Ah.. should have seen this earlier...
I connected the Raspi by wired LAN, but it had a WLAN configuration too and preferred the WLAN. So I was actually not connected to LAN but WLAN.
I deleted the WLAN PWD and now everything works...
Thank you for your support and the great software!

NickBuilder, 2025-02-06:
Glad to hear you got it working. 👍
Not my code though. I'm just another user.