ct-Open-Source/tuya-convert #1143: install_prereq.sh needs --break-system-packages on Debian 12 (or another environment)

Auravendill, 2024-11-12:
The install script tries to install python packages in the default environment, which is also the one apt installs to. Since Debian 12 this environment is a bit protected, so any pip install without explicit "--break-system-packages" will just fail.

This is the output, when it fails:

`0 aktualisiert, 0 neu installiert, 0 zu entfernen und 0 nicht aktualisiert.
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.

    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.`

ybungalobill, 2024-11-13:
Also happens for me on Arch (up-to-date at the time of writing).

sparkydave1981, 2024-11-17:
I just tried to install on a new machine and got the same issue.  Is there a work-around?

suchoi31, 2024-11-24:
Same issue here.
Typing "sudo nano /etc/pip.conf" will lead to the following message in the console (putty) "sudo nano /etc/pip.conf: command not found"

Hope anyone can help fixing this issue.

tlghndeniz, 2024-11-24:
> Same issue here. Typing "sudo nano /etc/pip.conf" will lead to the following message in the console (putty) "sudo nano /etc/pip.conf: command not found"
> 
> Hope anyone can help fixing this issue.

1. Check If nano Is Installed
Run the following command to check if nano is available:

`which nano`

- If it outputs a path (e.g., /usr/bin/nano), nano is installed, but there might be another issue.
- If no output is displayed, it means nano is not installed.

2. Install nano
If nano is not installed, install it using your system's package manager:

- For Debian/Ubuntu-based systems:
```
sudo apt update
sudo apt install nano
```

tylertylerday, 2024-12-12:
I used the break-system-packages method and at the end of installing requirements it gives a warning about "Running pip as the 'root' user can result in broken permissions...etc" 

Should I be worried about that?

mattuschek, 2025-02-01:
> The install script tries to install python packages in the default environment, which is also the one apt installs to. Since Debian 12 this environment is a bit protected, so any pip install without explicit "--break-system-packages" will just fail.

Solution/Lösung:
https://github.com/ct-Open-Source/tuya-convert/discussions/1149#discussioncomment-11709361