ct-Open-Source/tuya-convert #1153: AttributeError: module 'ssl' has no attribute 'wrap_socket'

kidexx, 2025-01-04:
I have the following issue:
In the log **smarthack-psk.log** I get
```
module 'ssl' has no attribute 'wrap_socket'
new client on port 443 from 10.42.42.33:60310
module 'ssl' has no attribute 'wrap_socket'
new client on port 443 from 10.42.42.33:60312
module 'ssl' has no attribute 'wrap_socket'
new client on port 443 from 10.42.42.33:60324
module 'ssl' has no attribute 'wrap_socket'
```

This seems to be related to change in python:
[https://github.com/eventlet/eventlet/issues/795](https://github.com/eventlet/eventlet/issues/795)

It is in the file psk-frontend.py

But I don't know how to change it to the new format.

haschler, 2025-01-13:
Just update the file inside the sslpsk library:
https://github.com/drbild/sslpsk/blob/e16c8662cf84942636f0e1eab264a6b0434c07b6/sslpsk/sslpsk.py

kidexx, 2025-01-19:
Thank you. 

I exchanged the file in /usr/local/lib/python3.12/dist-packages/sslpsk/ - but I still get an error:

`new client on port 443 from 10.42.42.33:49674
wrap_socket() got an unexpected keyword argument 'ssl_context'
new client on port 443 from 10.42.42.33:49676
wrap_socket() got an unexpected keyword argument 'ssl_context'
`