ct-Open-Source/tuya-convert #1157: new tuya smart plug 20A convert failed attempt

mabeoni, 2025-03-05:
hello, I have tried to OTA convert a recent 20A tuya smart plug. I opened the box to solder and had-tasmotize the board but, having not found the TX RX and GND pins I opted for this procedure.
I tried with a brand new not yet configured plug, connected to the mains, it started to blink (I think in the first power on it goes all the way to pairing mode). But this is the output of the start_flash script.
Any suggestion?
Thanks in advance.

[image]
[image]
[image]

./start_flash.sh
tuya-convert v2.4.6
Checking for network interface wlp4s0... Found.
Checking UDP port 53... Occupied by systemd-resolve with PID 495.
Port 53 is needed to resolve DNS queries
Do you wish to terminate systemd-resolve? [y/N] y
Attempting to stop systemd-resolved.service
Checking UDP port 67... Available.
Checking TCP port 80... Available.
Checking TCP port 443... Available.
Checking UDP port 6666... Available.
Checking UDP port 6667... Available.
Checking TCP port 1883... Available.
Checking TCP port 8886... Available.
======================================================
  Starting AP in a screen.
  Starting web server in a screen
  Starting Mosquitto in a screen
  Starting PSK frontend in a screen
  Starting Tuya Discovery in a screen

=====================================================

======================================================
Starting smart config pairing procedure
Waiting for the device to install the intermediate firmware
Put device in EZ config mode (blinking fast)
Sending SSID                  vtrust-flash
Sending wifiPassword
Sending token                 00000000
Sending secret                0101
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
..................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
.................
SmartConfig complete.
Resending SmartConfig Packets
............
Timed out while waiting for the device to (re)connect
======================================================
Attempting to diagnose the issue...
No ESP82xx based devices connected according to your wifi log.
Here is a list of all the MAC addresses that connected:
b2:8f:ff:b8:4c:10

ke-ss, 2025-03-17:
I have the same issue. I got as far as you did and no further. 
I have not opened mine. Were you able to identify the chip in use?
Is it a ESP-32-C, and ESP8266 or a RealTek chip which is apparentley currently unflashable becasue no one has done the work.
ty

ke-ss, 2025-03-17:
I was able to connect to the device as a hot spot with SSID "vtrust-flash", so I am wonderinhg if some sort of pre-flash worked.

KarlBenjaminHorn, 2025-06-09:
It's a Eswin ECR6600, you can read the name on the chip in the upper left corner of the photo. i have the same. Unfortunately i haven't found any OTA Firmware yet. But it doesn't work with this project .