ct-Open-Source/tuya-convert #1159: error: This environment is externally managed

ricardopretrazy, 2025-04-01:
I've followed the instructions, and when running the `install_prereq.sh` , but once it has finished installing the `apt-get` packages, it has an error as below.

    error: externally-managed-environment

    × This environment is externally managed
    ╰─> To install Python packages system-wide, try apt install
        python3-xyz, where xyz is the package you are trying to
        install.

I've copied the line out of the `.sh` file, and edited it from

`sudo python3 -m pip install --user --upgrade paho-mqtt tornado git+https://github.com/drbild/sslpsk.git pycryptodomex`

to 

`python3 -m pip install --break-system-packages --user --upgrade paho-mqtt tornado git+https://github.com/drbild/sslpsk.git pycryptodomex`

(i.e. removed `sudo` and added `--break-system-packages`)

This seems to install all of the Python packages correctly, but can the developers confirm that this is correct please??

PetrovskYYY, 2025-04-27:
Hello, you can create virtual environment and install required python packages there:
```bash
sudo su
apt install python3-venv
python3 -m venv venv
source venv/bin/activate
pip3 install paho-mqtt tornado git+https://github.com/drbild/sslpsk.git pycryptodomex
```
Then proceed to the configuration step and running a script:
```bash
nano config.txt
./start_flash.sh
```

dl476375, 2025-11-12:
I am getting the same error and did the above steps mentioned by @PetrovskYYY , still the same thing

TrabacchinLuigi, 2025-11-18:
same problem here

badmin-c, 2025-12-06:
> Hello, you can create virtual environment and install required python packages there:
> 
> sudo su
> apt install python3-venv
> python3 -m venv venv
> source venv/bin/activate
> pip3 install paho-mqtt tornado git+https://github.com/drbild/sslpsk.git pycryptodomex
> Then proceed to the configuration step and running a script:
> 
> nano config.txt
> ./start_flash.sh

If you do this, you have to edit some scripts as the system-wide python bin is hard-coded in them. 
You need to replace every occurrence of '/usr/bin/env python3' with the path to the python3 bin from your virtual environment.
To get the path just enter the environment and type 'which python3'

The scripts you need to edit are:

fake-registration-server.py 
mq_pub_15.py 
psk-frontend.py 
setup_ap.sh 
tuya-discovery.py 

If you get the module 'ssl' has no attribute 'wrap_socket' error after that, just follow this link: 
https://github.com/ct-Open-Source/tuya-convert/issues/1153#issuecomment-2588133766