ct-Open-Source/tuya-convert #1163: Add nix flake to documentation

SHU-red, 2025-06-13:
Hi there,
i think i made it happen to

- Create a nix flake satisfying all dependencies
- Deactivate NetworkManager and firewall (necessary) on entering flake devShell
- Re-Activate NetworkManager and firewall on leaving flake devShell

So using the flake below, you only need to run `nix develop` in the same directory

Then follow the official documentation (clone git, edit config, run `start_flash_sh`) and if you leave the flake, everything gets re-enabled

Maybe thats convenient for others

```
{
  description = "Dev shell with Python, MQTT, sslpsk, etc.";

  inputs = {
    nixpkgs.url = "github:NixOS/nixpkgs/nixos-unstable";
    flake-utils.url = "github:numtide/flake-utils";
  };

  outputs = { self, nixpkgs, flake-utils, ... }:
    flake-utils.lib.eachDefaultSystem (system:
      let
        pkgs = import nixpkgs {
          inherit system;
          config.allowUnfree = true;
        };

        python = pkgs.python3;

        pythonEnv = python.withPackages (ps: with ps; [
          paho-mqtt
          tornado
          pycryptodomex
          (ps.buildPythonPackage {
            pname = "sslpsk";
            version = "unstable-2021-05-28";
            src = pkgs.fetchFromGitHub {
              owner = "drbild";
              repo = "sslpsk";
              rev = "d88123a75786953f82f5e25d6c43d9d9259acb62";
              sha256 = "sha256-RqaZLtRMzYJPKXBBsw1alujGyqWAQRSQLPyAR8Zi6t4=";
            };
            nativeBuildInputs = [ pkgs.openssl.dev ];
            propagatedBuildInputs = [ pkgs.openssl ];
            pythonImportsCheck = [ "sslpsk" ];
          })
        ]);

      in {
        devShells.default = pkgs.mkShell {
          name = "prerequisites-shell";

          buildInputs = [
            pkgs.git
            pkgs.iw
            pkgs.dnsmasq
            pkgs.util-linux
            pkgs.hostapd
            pkgs.screen
            pkgs.curl
            pkgs.mosquitto
            pkgs.haveged
            pkgs.nettools
            pkgs.openssl.dev
            pkgs.openssl
            pkgs.iproute2
            pkgs.iputils
            pythonEnv
          ];

          shellHook = ''
            echo "✅ Development environment ready."
            echo "Deactivating firewall and NetworkManager services..."
            sudo systemctl stop NetworkManager.service
            sudo systemctl stop firewall

            cleanup() {
              echo "🔄 Re-activating firewall and NetworkManager..."
              sudo systemctl start firewall
              sudo systemctl start NetworkManager.service
            }

            trap cleanup EXIT
          '';
        };
      });
}
```

mberndt123, 2025-08-30:
Doesn't work for me.
```
warning: Git tree '/home/mberndt/tuya-convert' is dirty
warning: creating lock file '"/home/mberndt/tuya-convert/flake.lock"': 
• Added input 'flake-utils':
    'github:numtide/flake-utils/11707dc2f618dd54ca8739b309ec4fc024de578b?narHash=sha256-l0KFg5HjrsfsO/JpG%2Br7fRrqm12kzFHyUHqHCVpMMbI%3D' (2024-11-13)
• Added input 'flake-utils/systems':
    'github:nix-systems/default/da67096a3b9bf56a91d16901293e51ba5b49a27e?narHash=sha256-Vy1rq5AaRuLzOxct8nz4T6wlgyUR7zLU309k9mBC768%3D' (2023-04-09)
• Added input 'nixpkgs':
    'github:NixOS/nixpkgs/dfb2f12e899db4876308eba6d93455ab7da304cd?narHash=sha256-1wxxznpW2CKvI9VdniaUnTT2Os6rdRJcRUf65ZK9OtE%3D' (2025-08-28)
warning: Git tree '/home/mberndt/tuya-convert' is dirty
error:
       … while calling the 'derivationStrict' builtin
         at <nix/derivation-internal.nix>:37:12:
           36|
           37|   strict = derivationStrict drvAttrs;
             |            ^
           38|

       … while evaluating derivation 'prerequisites-shell'
         whose name attribute is located at /nix/store/4hm8lf740i8qvyg5pzdqfm0rpshwb7vn-source/pkgs/stdenv/generic/make-derivation.nix:538:13

       … while evaluating attribute 'buildInputs' of derivation 'prerequisites-shell'
         at /nix/store/4hm8lf740i8qvyg5pzdqfm0rpshwb7vn-source/pkgs/stdenv/generic/make-derivation.nix:590:13:
          589|             depsHostHost = elemAt (elemAt dependencies 1) 0;
          590|             buildInputs = elemAt (elemAt dependencies 1) 1;
             |             ^
          591|             depsTargetTarget = elemAt (elemAt dependencies 2) 0;

       (stack trace truncated; use '--show-trace' to show the full, detailed trace)

       error: python3.13-sslpsk-unstable-2021-05-28 does not configure a `format`. To build with setuptools as before, set `pyproject = true` and `build-system = [ setuptools ]`.`
```

mberndt123, 2025-08-30:
> If you add what the flake tells you to add then it works

Yes, I tried that later and ran into another, specifically #1153. I figured it's probably due to a new OpenSSL version, so I randomly tried an older NixOS version: `nixpkgs.url = "github:NixOS/nixpkgs/22.05";`, and that made the error go away. Unfortunately, it didn't make a difference. While `smarthack-psk-log` no longer has critical errors, it still won't install Tasmota on the Wifi smart switch I bought on AliExpress :-(
At this point I suspect it's simply not compatible, it appears to be an ESP32-C3 rather than an ESP8266.